TypQxQ/Sigenergy-Local-Modbus #8: Migration from old yaml code

TypQxQ, 2025-03-22:
I've used the "old" integration since day one, and I have become very addicted to all the nice data it gives me. :-)
I will of course adapt to your new shiny HACS integration and continue to have full control over my Sigenergy plant.
But I have one concern about the old data, do I loose it when I remove the old integration? Is there a way to keep it in the HA database?
I rader not loose six months of recorded data! :-)

_Originally posted by @Filialen in https://github.com/TypQxQ/HACS-Sigenergy-Local-Modbus/discussions/5#discussioncomment-12586556_

TypQxQ, 2025-03-22:
> Maybe, it is a way to rename the new entities with the old names. Then, the old values should remain.
> 
> (However, it is would be better to change all old YAML entities which should remain to the new entity names.)

That doesn't work because HA uses meta entities to identify the entities in the history for the history to persist even though you change entity id.
From my research the only way is to change directly in the database risking corruption of the whole database.
I will try to initialise calculated values.

Gloomyeye, 2025-03-22:
ah okay. Yes, it would be great if the accumulated values can be intialized with a start value (or if the calculated values differs to much to calibrate the value, e.g. like a utility sensor.)

cjungde, 2025-03-24:
I am using InfluxDB and Grafana inside HA in addition to the old, manual Sigenergy installation.
Any idea chance to keep that data as well ?

TypQxQ, 2025-04-17:
Added in v0.3.7

closed 2025-04-17 by TypQxQ

TypQxQ, 2025-04-18:
> Thanks! I can confirm that the migration works. Maybe it would be good to add a functionality (e.g. action) to calibrate the accumulated sensors if these differs to much from the sigen app values.

https://github.com/TypQxQ/HACS-Sigenergy-Local-Modbus/discussions/39