TypQxQ/Sigenergy-Local-Modbus #9: Logs after install

johnf, 2025-03-22:
I noticed the following entries in the logs after setting this up. Not sure if they are good or bad :)

```
2025-03-23 10:15:46.206 WARNING (MainThread) [custom_components.sigen.sensor] Skipping integration sensor Sigen Plant Accumulated PV Energy because source entity with key plant_photovoltaic_power not found
2025-03-23 10:15:46.206 WARNING (MainThread) [custom_components.sigen.sensor] Skipping integration sensor Sigen Plant Accumulated Grid Export Energy because source entity with key plant_grid_export_power not found
2025-03-23 10:15:46.206 WARNING (MainThread) [custom_components.sigen.sensor] Skipping integration sensor Sigen Plant Accumulated Grid Import Energy because source entity with key plant_grid_import_power not found
2025-03-23 10:15:46.206 WARNING (MainThread) [custom_components.sigen.sensor] Skipping integration sensor Sigen Plant Daily Grid Export Energy because source entity with key plant_grid_export_power not found
2025-03-23 10:15:46.206 WARNING (MainThread) [custom_components.sigen.sensor] Skipping integration sensor Sigen Plant Daily Grid Import Energy because source entity with key plant_grid_import_power not found
2025-03-23 10:15:46.207 WARNING (MainThread) [custom_components.sigen.sensor] Skipping integration sensor Sigen Plant Daily PV Energy because source entity with key plant_photovoltaic_power not found
2025-03-23 10:15:46.207 WARNING (MainThread) [custom_components.sigen.sensor] Skipping integration sensor Sigen Plant Accumulated Consumed Energy because source entity with key plant_consumed_power not found
2025-03-23 10:15:46.207 WARNING (MainThread) [custom_components.sigen.sensor] Skipping integration sensor Sigen Plant Daily Consumed Energy because source entity with key plant_consumed_power not found
2025-03-23 10:15:46.208 WARNING (MainThread) [custom_components.sigen.sensor] Skipping integration sensor Sigen Inverter Accumulated PV Energy because source entity with key inverter_pv_power not found

2025-03-23 10:16:29.631 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'sigen' sets option flow config_entry explicitly, which is deprecated at custom_components/sigen/config_flow.py, line 566: self.config_entry = config_entry. This will stop working in Home Assistant 2025.12, please create a bug report at https://github.com/custom-components/sigen/issues

2025-03-23 10:20:10.386 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.sigen_plant_phase_a_reactive_power (kVar) cannot be converted to the unit of previously compiled statistics (kVAr). Generation of long term statistics will be suppressed unless the unit changes back to kVAr or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2025-03-23 10:20:10.386 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.sigen_plant_phase_b_reactive_power (kVar) cannot be converted to the unit of previously compiled statistics (kVAr). Generation of long term statistics will be suppressed unless the unit changes back to kVAr or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2025-03-23 10:20:10.386 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.sigen_plant_phase_c_reactive_power (kVar) cannot be converted to the unit of previously compiled statistics (kVAr). Generation of long term statistics will be suppressed unless the unit changes back to kVAr or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
```

I have a SigenStor with
* 3 x 8kWh batteries
* 25kW DC Charger
* 15kW 3 Phase Inverter

TypQxQ, 2025-03-23:
Thank you, please check if they occur again after a restart.

johnf, 2025-03-23:
The unit conversion has gone away. I did see a repair entry in HA, though, and I told it to delete the old data. So I suspect this was due to the old way and the new way.

Should the DC charger module have been detected? Happy to do some debugging to work out why on my end and do a PR if needed

johnf, 2025-03-23:
Oh and the other errors haven't reappeared

TypQxQ, 2025-03-23:
> The unit conversion has gone away. I did see a repair entry in HA, though, and I told it to delete the old data. So I suspect this was due to the old way and the new way.
> 
> Should the DC charger module have been detected? Happy to do some debugging to work out why on my end and do a PR if needed

You have to add it manually.

johnf, 2025-03-23:
I get his when I try that

[image]

But I already have an inverter

[image]

At some stage during me looking and trying to add this message popped up again

```
 2025-03-23 16:36:28.346 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'sigen' sets option flow config_entry explicitly, which is deprecated at custom_components/sigen/config_flow.py, line 566: self.config_entry = config_entry. This will stop working in Home Assistant 2025.12, please create a bug report at https://github.com/custom-components/sigen/issues

```

TypQxQ, 2025-03-23:
I'm rewriting the installation and configuration part. Will debug this after that.

johnf, 2025-04-12:
Adding DC charger works on latest version

closed 2025-04-12 by johnf

TypQxQ, 2025-04-12:
Fantastic news!
Thank you for testing.

Could you at some point also check please check if the switch in v. 0.3.4 is working for DC Charger too.
Also, would be great with a screenshot of it working 😄 for later documentation.