TypQxQ/Sigenergy-Local-Modbus #15: HA Crashing Out of Memory

fbradyirl, 2025-03-29:
I've noticed since installing the integration that HA is crashing pretty regularly with out of memory. 

Logs attached: 

[ha crash logs investigation_with_debug.log](https://github.com/user-attachments/files/19521146/ha.crash.logs.investigation_with_debug.log)

I can't be 100% sure this is caused by this integration.

I deleted my DB yesterday and already it is back at 2.5gb in size. Again, not 100% sure if this integration is causing this but I havent added any other new integrations the last while.

```
[core-ssh ~]$ ha core info
arch: aarch64
audio_input: null
audio_output: null
backups_exclude_database: false
boot: true
image: ghcr.io/home-assistant/yellow-homeassistant
ip_address: 172.30.32.1
machine: yellow
port: 8123
ssl: false
update_available: false
version: 2025.3.4
version_latest: 2025.3.4
watchdog: false
[core-ssh ~]$
```

TypQxQ, 2025-03-30:
I can't see this problem on my dev system. The DB is not as large.
[image]
If you share your db I can try debugging it.

fbradyirl, 2025-03-30:
I’ve cleared the DB again and that size issue seems to be gone now. Could be something else causing that.

I’ve had another crash. 

```
[core-ssh config]$ dmesg | grep -i 'killed process'
[271467.179601] Out of memory: Killed process 2029749 (influxd) total-vm:1217216kB, anon-rss:182144kB, file-rss:6016kB, shmem-rss:0kB, UID:0 pgtables:1488kB oom_score_adj:200
[271504.492263] Out of memory: Killed process 2030614 (influxd) total-vm:1217344kB, anon-rss:178944kB, file-rss:8320kB, shmem-rss:0kB, UID:0 pgtables:1480kB oom_score_adj:200
[271529.735736] Out of memory: Killed process 2031117 (python3) total-vm:153560kB, anon-rss:124068kB, file-rss:3712kB, shmem-rss:0kB, UID:0 pgtables:340kB oom_score_adj:200
[271540.462434] Out of memory: Killed process 2031224 (influxd) total-vm:1217600kB, anon-rss:178688kB, file-rss:7808kB, shmem-rss:0kB, UID:0 pgtables:1516kB oom_score_adj:200
[271568.987916] Out of memory: Killed process 1938888 (python3) total-vm:4186156kB, anon-rss:2572832kB, file-rss:3072kB, shmem-rss:0kB, UID:0 pgtables:7328kB oom_score_adj:-300
[core-ssh config]$ 

```

Seems to be happening during some calculated conversion in the integration. 

```

2025-03-30 12:21:45.686 DEBUG (MainThread) [custom_components.sigen.calculated_sensor] [CS][State] Scheduling check for sensor.sigen_plant_accumulated_grid_export_energy - max_interval: 0:00:30, source_state: 2.184
2025-03-30 12:21:45.686 DEBUG (MainThread) [custom_components.sigen.calculated_sensor] [CS][State] Converted 2.184 to Decimal: 2.184
2025-03-30 12:21:45.686 DEBUG (MainThread) [custom_components.sigen.calculated_sensor] [CS][Integration] Max interval exceeded for sensor.sigen_plant_daily_grid_export_energy at 2025-03-30 11:21:45.686547+00:00
2025-03-30 12:21:45.686 DEBUG (MainThread) [custom_components.sigen.calculated_sensor] [CS][Integration] Time-based calculation - Elapsed: 0.03928299999999999847499765337488497607409954071044921875 seconds, State: 2.184
2025-03-30 12:21:45.687 DEBUG (MainThread) [custom_components.sigen.calculated_sensor] area = 0.08579407199999999666939487497, area_scaled = 0.00002383168666666666574149857638 new state = 8.825576794397500000098027837
2025-03-30 12:21:45.687 DEBUG (MainThread) [custom_components.sigen.calculated_sensor] [CS][State] sensor.sigen_plant_daily_grid_export_energy value: 8.825577 (type: Decimal)
2025-03-30 12:21:45.687 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.sigen_plant_daily_grid_export_energy, old_state=<state sensor.sigen_plant_daily_grid_export_energy=8.825553; state_class=total, source_entity=sensor.sigen_plant_grid_export_power, unit_of_measurement=kWh, device_class=energy, friendly_name=Sigen Plant Daily Grid Export Energy @ 2025-03-30T12:21:45.646640+01:00>, new_state=<state sensor.sigen_plant_daily_grid_export_energy=8.825577; state_class=total, source_entity=sensor.sigen_plant_grid_export_power, unit_of_measurement=kWh, device_class=energy, friendly_name=Sigen Plant Daily Grid Export Energy @ 2025-03-30T12:21:45.687336+01:00>>
2025-03-30 12:21:45.687 DEBUG (MainThread) [custom_components.sigen.calculated_sensor] [CS][State] Scheduling check for sensor.sigen_plant_daily_grid_export_energy - max_interval: 0:00:30, source_state: 2.184
2025-03-30 12:21:45.688 DEBUG (MainThread) [custom_components.sigen.calculated_sensor] [CS][State] Converted 2.184 to Decimal: 2.184
[11:50:09] INFO: Home Assistant Core finish process exit code 256
[11:50:09] INFO: Home Assistant Core finish process received signal 9
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

I noticed the power state of the invert changed right at the same time (as i have a notification to iOS when that happens). Not sure if that is related.

fbradyirl, 2025-03-30:
Note i have debug logs enabled for everything:

fbradyirl, 2025-03-30:
I’m starting to think it might be the VSCode addon causing the trouble as i see multiple issues [like this](https://github.com/hassio-addons/addon-vscode/issues/565) on their repo.

fbradyirl, 2025-04-01:
```
2025-04-01 15:15:01.166 ERROR (MainThread) [homeassistant.components.recorder.core] The recorder backlog queue reached the maximum size of 2100787 events; usually, the system is CPU bound, I/O bound, or the database is corrupt due to a disk problem; The recorder will stop recording events to avoid running out of memory
[14:15:26] INFO: Home Assistant Core finish process exit code 256
[14:15:26] INFO: Home Assistant Core finish process received signal 9
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
[core-ssh ~]$
[core-ssh ~]$
[core-ssh ~]$ dmesg | grep -i 'killed process'
[59894.254332] Out of memory: Killed process 5590 (python3) total-vm:5860420kB, anon-rss:2804452kB, file-rss:5860kB, shmem-rss:0kB, UID:0 pgtables:10196kB oom_score_adj:-300
[core-ssh ~]$
```
So I uninstalled the VSCode addon and am seeing the core crash again. 

I'm thinking one of the HACS sensors is writing too much to the recording service which overloaded the system. Worth a discussion...

fbradyirl, 2025-04-02:
I've shared my DB with you via email @TypQxQ

TypQxQ, 2025-04-03:
Looks as your integration has a scan interval that is way to high, less than a second. 
DEFAULT_SCAN_INTERVAL = 5
It is in seconds and I can't recreate it now. I've added a debug log in the latest dev branch but other things are not working there yet.

fbradyirl, 2025-04-04:
> Looks as your integration has a scan interval that is way to high, less than a second.

What that something I did in my own config?

Since I uninstalled the integration a couple of days ago, the problem is gone.

tobiaso88, 2025-04-08:
I've also hade problem with HA crashing every day around 12:00.
Now, after i removed the integration the problem is gone.

TypQxQ, 2025-04-11:
Please check and report on v.0.3.1

TypQxQ, 2025-04-14:
Please try v0.3.4
I did change lots of code for the integrated sensors to check so they do not atempt to save states more often than once every time a modbus request is made.

My DB is only 13MB 

You can delete the device in HA, then in "Developer Tools" go to Statistics and hopefully you can delete the sensors there.

/Andrei

ksya, 2025-04-14:
I think it does not work. Now the sensors get an update about once a second, while the polling interval is set to 10 seconds.. And again this interval seems to be decreasing over time, because 5 minutes ago, it was refreshing every 5 seconds.

TypQxQ, 2025-04-15:
Hi!
Please check v0.3.6

Thank you for your continuing support!
I have rewritten the timers for the integration sensors and have run it several hours without problems yet.
Also checked the databas state updates are now no more frequent than the set frequency.

/Andrei

ksya, 2025-04-15:
Looks good now, thanks!

closed 2025-04-15 by TypQxQ