TypQxQ/Sigenergy-Local-Modbus #18: EVAC Set AMPs fails with exception

fbradyirl, 2025-03-31:
Changing EVAC Amps results in: 

` Failed to set value 23.0 for Sigen AC Charger Charger Output Current: Error writing registers: Error writing registers at address 42001: ExceptionResponse(dev_id=2, transaction_id=28665, address=0, count=0, bits=[], registers=[], status=1)`

```
Logger: custom_components.sigen.number
Source: custom_components/sigen/number.py:663
```
and the value does not change.

TypQxQ, 2025-04-11:
Hi!
Please try again with the new 0.3.0 version or higher.

fbradyirl, 2025-04-16:
Tried but still not working I'm afraid.

TypQxQ, 2025-04-17:
But you can read the values right?
Can you please provide me with the settings for all your Sigen devices?
Host, Port, DeviceID

TypQxQ, 2025-04-17:
Please also try the new V0.3.8

mattgrech, 2025-04-22:
Hey @TypQxQ thanks for all the hard work you are doing on this integration. I've finally been able to read the values for the AC charger in your latest 3.11 version using inverter as slave 2 and ac charger as slave 1.

I can't however write any values for the AC charger.

This is the error when trying to turn on/off the ac charger

[image]

This is the error when adjusting the amps

[image]

mattgrech, 2025-04-22:
Also @TypQxQ let me know if you need further details of the system to troubleshoot

TypQxQ, 2025-04-22:
I'll reach out to Sigenergy. This should work the same as for the inverter.
You do receive state updates from it I asume.

/Andrei

TypQxQ, 2025-04-23:
Can you please ensure that OCPP is **DISABLED** and that charger is not in fault situation, meaning charger is not plugged in, ac charger self failure etc., What is your AC-Charger System State?

Then please check if start/stop works.

What value are you trying to change to in "Charger Output Current"? Must be between 6 and Rated Current.

/Andrei

mattgrech, 2025-04-23:
No faults
[image]

OCPP is disabled
[image]

Charger is not plugged in
[image]

Changing to any value gives an error but in this example i tried changing to 13

[image]

Not turning the charger off also gives an error and i note its always on

[image]

TypQxQ, 2025-04-23:
Try the on switch again after plugging in.

mattgrech, 2025-04-23:
Ok confirming the switch (start/stop) works when plugged in. Noting the state doesn't seem to change in the integration along with time it changed

[image]

Changing amps still results in an error and when the error is caused all values for the sig integration show unknown until integration is reset

TypQxQ, 2025-04-23:
Please check if you get any errors in the logs.
Does the AC-Charger System State change?

mattgrech, 2025-04-23:
@TypQxQ System State does not change. Here are the logs

[image]

[image]

[image]

TypQxQ, 2025-04-23:
Hello!
Made som tweaks to logging during lunch and have now updated to v1.0.3.
Please provide a log with the new version, as described here: https://github.com/TypQxQ/Sigenergy-Local-Modbus/discussions/54

fbradyirl, 2025-04-23:
Hi @TypQxQ i finally got my slave working for the AC charger again. I have emailed you debug logs when trying to change amps (better to email as i dont want my logs public)…

fbradyirl, 2025-04-23:
Also note that the ac charger system state is not numeric in my case.

[image]

So the logic to determine if charging or not will need updating.

```python
  SigenergySwitchEntityDescription(
        key="ac_charger_start_stop",
        name="AC Charger Power",
        icon="mdi:ev-station",
        # identifier here will be ac_charger_name
        is_on_fn=lambda data, identifier: data["ac_chargers"].get(identifier, {}).get("ac_charger_system_state") > 0,
        turn_on_fn=lambda coordinator, identifier: coordinator.async_write_parameter("ac_charger", identifier, "ac_charger_start_stop", 0),
        turn_off_fn=lambda coordinator, identifier: coordinator.async_write_parameter("ac_charger", identifier, "ac_charger_start_stop", 1),
    ),
```

When value is `B1`, my car is plugged in but not charging, It is “reserving” and waiting for scheduled charge later tonight.

[image]

fbradyirl, 2025-04-23:
If i trigger manual ‘Start’ in the app, it goes to B2 for a second but settling on C2.

[image]

So I guess

* B1: Plugged in, reserving
* B2: Starting charge?
* C2: Charging

mattgrech, 2025-04-23:
@TypQxQ i am getting an error when downloading the logs.

500 Internal Server Error
Server got itself in trouble

Hope the logs from @fbradyirl work and is enough for you?

TypQxQ, 2025-04-24:
Thank you for the logs.
I've done a change to the code, please see if this helps.

mattgrech, 2025-04-24:
@TypQxQ looks like we have progress with changing amps :) the amps correctly go up and down and i can see the change in the tesla app but interestingly the amps doesn't change in the Sigen App but i can live with that.

The AC charger Power switch does work but it always shows as "on" even though its actually off, is that something to do with the AC Charger state mapping?

mattgrech, 2025-04-24:
I have successfully charged the car this morning with the new integration and was able to change the amps on the fly so great work @TypQxQ 

These are my take of definitions of Sigen AC Charger System State based on testing

A1/A2 - Available (Not plugged in)
B1 - Plugged in (Reserving)
B2 - Preparing
C2 - Charging

It would be great if we could make an update or create another sensor to reflect these definitions?

Also the following switch isn't acting the way a switch normally works

switch.sigen_ac_charger_ac_charger_power

It is always showing as "On" if you press "off" it will change to grey and move to the off position but the state says "On". Refreshing the page will then show the switch as "On"

TypQxQ, 2025-04-25:
Thank you.

I'm very happy with the progress. 

I will take a look when I have some free time.
Been busy with work and the kids yesterday.

cjungde, 2025-04-27:
Seems here is the full list of states:
https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/discussions/2#discussioncomment-12137854

TypQxQ, 2025-04-27:
New 105 version should have better description of states. and the on off button should work better.

mattgrech, 2025-04-27:
@TypQxQ thanks for the updates, description states are looking good. Unfortunately still the same results for the on/off button

ronanohbee, 2025-05-02:
The integration is working for plant, inverter abs PVs but does not find the EVAC.  Is a particular installer setup required for this too?

cjungde, 2025-05-07:
Short feedback. The EVAC modbus integration works very well.  

[image]

[image]

TypQxQ, 2025-05-13:
Thank you evertone for help with this issue.

closed 2025-05-13 by TypQxQ

fbradyirl, 2025-05-13:
Is set amps fixed for everyone? For me, when I set amps in HA, it has no effect when I check the MySigen app. Debug logs emailed to you just now.

mattgrech, 2025-05-13:
On/Off still remains buggy

TypQxQ, 2025-05-14:
> Is set amps fixed for everyone? For me, when I set amps in HA, it has no effect when I check the MySigen app. Debug logs emailed to you just now. 

From what I've understood you have to connect the car and maybe start charging before changing amps and it will bot show up as changed in the app but reflected in the car dashboard. Bug is reported to Sigenergy that it doesn't show in the app but may be fixed in v3 of the app.

TypQxQ, 2025-05-14:
> On/Off still remains buggy

Can you please open a new issue so we keep it separated from the set amps. Also specify what doesn't work. I don't have a Sigenergy Charger and no way of testing it myself.