TypQxQ/Sigenergy-Local-Modbus #19: how can the default scan-interval be changed?

holli73, 2025-04-04:
i have one sigen-gateway + 2x 25kw hybrid inverter with 48kw battery each  - i changed the default value in const.py but still the refresh is way to often i believe ( 5 seconds) - because if i enable this integration - it takes just a couple of minutes that the entities become unavailable and after some time they come back and if this runs longer the interval between showing and not showing values it getting longer and longer - it looks to me that if both inverters are queried that 5 seconds is just a too small interval - i even changed this in the old (package version) to 30 seconds.

any hint would be great where i can change the scan-interval to verify this was/is the root cause and the delay is just caused because of the 3 ip's my sigen env. has. - i publish the values on a ha-floorplan - looks like this:

[image]

thanks
holli

PS: the GRID CODE for AUSTRIA is 26 maybe this can be added as well.

fbradyirl, 2025-04-04:
I noticed the same thing. For many hours the entities were unavailable and then suddenly available again. I wonder is there a rate limit on Modbus where it shuts off the server if too many requests were made.

TypQxQ, 2025-04-04:
I've heard there is a limit of 1 req per second. 

I have to finish the rewritten version and maybe it will be fixed. I have made quite a lot of changes.  Have limited time though. I'm away with my daughter who's competing in the Jiujitsu nationals tomorrow so no progress done today.

holli73, 2025-04-11:
@TypQxQ - i did use the latest release and all looks ok so far - as i have 2x 25kw inverters - i was able to add the 2nd one with its ip and modbus id - really cool 

thanks a lot 
holli

PS: i noticed that the sensors for the plant have a prefix - but the found inverters do not have a "sigen" prefix anymore?

[image]

TypQxQ, 2025-04-11:
Good catch with the naming inconsistancy.
I fixed it now in 0.3.0

TypQxQ, 2025-04-11:
Also you can change scan interval now in plant configuration.

holli73, 2025-04-11:
@TypQxQ - it looks like if the interval is changed - it is not taken - i changed it to 33 seconds but in the log i still can see
```
2025-04-11 22:14:32.854 DEBUG (MainThread) [custom_components.sigen] Finished fetching sigen_192.168.245.33_502 data in 1.564 seconds (success: True)
2025-04-11 22:14:38.147 DEBUG (MainThread) [custom_components.sigen] Finished fetching sigen_192.168.245.33_502 data in 1.859 seconds (success: True)
```
every time the configuration is opened again the slider is back to  5 seconds - my second inverter stopped the modbus again - so it still hits the system too often - i would guess.

thanks
holli

TypQxQ, 2025-04-11:
Thank you!
Check new v0.3.2. 
Should be fixed now.
/Andrei

closed 2025-04-15 by TypQxQ

holli73, 2025-04-15:
@TypQxQ 
i'm still on 0.3.3 and did lots of testing and was able to get it stable running - my sigen env is connected only through a single cable from the sigen-gateway to my network -> both sigen-store are connected through network cable to the sigen-gateway - the system has 3 ip's (gw + 2 inverter) - on the installer app by default the modbus is set '1' for both inverter and that is the reason i used the 2 different ip's - but this got never stable as inverter 2 caused issues and both got unavailable - since we changed inverter 2 to a different modbus id - and i use only the ip for inverter#1 for both - i did not have any issue yet - as you can see from the screenshot in the past it was not stable - but now it looks ok - the config options are not usable at least in 0.3.3 as only one IP is needed - if the modbus settings are correctly done for each inverter 

- configure-plant -> doesn't need an ip / port only scan-interval (or just the first inverter = plant)
- configure-inverter needs ip + port + modbus id but only if it is the first or is inherited by the plant config
- any further inverter should use the same ip / port and only a different modbus id is needed.

f.ex. GW = 192.168.245.22 / IV1 = 192.168.245.33 / IV2 = 192.168.245.34 and i use following settings now -> 192.168.245.33 502 1 for IV1 and 192.168.245.33 502 2 for IV2 and did not have any drops yet with a scan interval of 33 seconds.

hope this makes sense - i'm not sure if i should go for the latest release yet - as i have it running stable for the first time **gg**

thanks a lot for your help and time
best regards
holli

[image]

TypQxQ, 2025-04-15:
Thank you for the feedback!
I will have it in the documentation when it comes to that.
I definitly recomend upgrading to at least 0.3.6 where I hope to have fixed bugs in the state updates where the Home Assistant database was floded with too many updates.

Also from 0.3.7 the modbus update will only update high frequency sensors every time, having selectable update intervals in 4 levels:
High, Alarm, Medium(V and Reactive Power) and Low (Serial nr, Firmware, etc.)

This means you can increase update time for important registers.

TypQxQ, 2025-04-15:
Would you be able to post screenshot of Modbus settings for first and second inverter?

holli73, 2025-04-15:
[image]
[image]

TypQxQ, 2025-04-15:
Thank you!
So no need to set the IP of the main inverter on the second, just connect to first with diffrent deviceID and go?
Easy...

holli73, 2025-04-15:
@TypQxQ correct only the IP of the first inverter is needed that's it and for the other only a unique ID is needed - because in the default installation this is set to 1 and modbus is disabled - this did not work that is the reason i thought - i need to set the ip of the second one - but this causes issues - since i changed both inverters to use the the ip of the first but just with a different modbus id this works fine and i had not any UNAVAILABLE since than - i updated to the latest as well and this shows now:

[image]

but i'm not sure if i should/would add this discovered devices as i have already both inverters and what will be shown from the gw (22)?

[image]

if i add this - will this mess up the current found/setup devices?

thanks
holli