TypQxQ/Sigenergy-Local-Modbus #33: Some sensors "unknown" (eg sigen_plant_accumulated_consumed_energy)

dennisroethig, 2025-04-17:
Hi, first of all a massive thank you for this project ♥️

I just moved from the yaml config to the HACS version. Most things seem to work but when I started to configure the Energy dashboard I noticed that some values just appear as "unknown"

[image]
[image]

I spotted those being used in an issue you resolved 2 days ago, so I assume they are the ones meant to be used? 😅

dennisroethig, 2025-04-17:
Something that just came to mind, it looks like the Sigenergy support set my SigenStor to the SPC107 version after the 108 one didn't install. I wonder if that could be related 🤔

(I reached out to them again to understand what has been happening there)

[image]

dennisroethig, 2025-04-17:
Just looked around and it sounds like it's similar to https://github.com/TypQxQ/HACS-Sigenergy-Local-Modbus/issues/11 maybe?

TypQxQ, 2025-04-17:
All "Accumulated" values are calculated by the integration from other values recieved by Modbus.
This has nothing to do with firmware but with this software.

Can you check if you have any messages in the error logs?

dennisroethig, 2025-04-17:
Awesome 🙌 Case closed 😊

closed 2025-04-17 by dennisroethig