TypQxQ/Sigenergy-Local-Modbus #62: ome of the Sigen Plant Accumulated sensors drop to zero.

TypQxQ, 2025-05-01:
Ok so I have determined that some event causes some of the Sigen Plant Accumulated sensors to drop to zero.

[image]

and from checking my HA logs it seems to do it upon HA restart.

[image]

So something on restart caused them to drop down. I think I did a supervisor (whole device) reboot at that time.

_Originally posted by @fbradyirl in https://github.com/TypQxQ/Sigenergy-Local-Modbus/discussions/59#discussioncomment-12990825_

Wartopia, 2025-05-01:
[home-assistant_sigen_2025-05-01T07-55-53.297Z.zip](https://github.com/user-attachments/files/19997720/home-assistant_sigen_2025-05-01T07-55-53.297Z.zip)

Ill upload one later with a reboot. Maybe you find something in this already.

fbradyirl, 2025-05-01:
ill have to wait until it happens again and will email you the logs then. Enabled debug logging just now.

Wartopia, 2025-05-01:
Debug restarts after reboot, no data from before. So only boot data in this one.

Reset after reboot: 0

Sigen Plant Accumulated Consumed Energy
Accumulated Grid Export Energy
Accumulated Grid Import Energy

Negativ stats: Not sure if this is ok, consumed cant/should not be negative right?

Sigen Plant Daily Consumed Energy
Sigen Plant Consumed Power
Daily Consumed Energy

[home-assistant_sigen_2025-05-01T08-42-01.170Z.log.zip](https://github.com/user-attachments/files/19998265/home-assistant_sigen_2025-05-01T08-42-01.170Z.log.zip)

michaelroper, 2025-05-02:
Just saw this thread, and checked my accumulated sensors - these ones all dropped to zero when I restarted HA to update the Sigen integration. Not sure if it was because of the restart or the update..

[image]

TypQxQ, 2025-05-02:
I've reproduced this and am trying to find the problem. Strange that it is not affecting the daily sensors....

Wartopia, 2025-05-03:
> I've reproduced this and am trying to find the problem. Strange that it is not affecting the daily sensors....

True, but as you could see, i get strange first hour stats from the last day in the energy dashboard from the daily ones.

So also there is something wrong. Maybe the reset is to late for those, could happen in the first second of the new day and not in the second of the last day. Could explain it.

Also someone else opend a new issue about the negative stats. So this also something separate going on.

Also, since not al stats from "Accumulated" are reset, i guess some are taken by modbus and some are calculated. My guess is the calculated reset? 

Could be a option you need to use to keep the stats, like history function etc.

When i create a helper function for counting or other stuff, i get the option to reset to zero with a reboot or keep the value.

TypQxQ, 2025-05-03:
Fixed in v.1.0.6
The states were saved but divided by 1000. This was caused by the calculated Accumulated being MWh and even though the last saved state is also in MWh, it converted it to kWh and showed up as MWh, so I multiply it by 1000 at loading if MWh.
Have to search if there is a better way in HomeAssistant for this. This is Home Assistant related in how it stores data.

closed 2025-05-03 by TypQxQ