TypQxQ/Sigenergy-Local-Modbus #64: Warning of calculated_sensor

Gloomyeye, 2025-05-02:
I got the following warning:

```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.sigen.calculated_sensor
Quelle: custom_components/sigen/calculated_sensor.py:346
Integration: Sigenergy ESS (Dokumentation, Probleme)
Erstmals aufgetreten: 09:40:05 (117 Vorkommnisse)
Zuletzt protokolliert: 11:53:45

[CS][Plant Consumed] Calculated power is negative: -0.06099999999999994 kW
[CS][Plant Consumed] Calculated power is negative: -1.1190000000000007 kW
[CS][Plant Consumed] Calculated power is negative: -0.007000000000000561 kW
[CS][Plant Consumed] Calculated power is negative: -0.06500000000000039 kW
[CS][Plant Consumed] Calculated power is negative: -0.15799999999999947 kW
```

I think that is normal? Is the warning only for debugging, if yes, then the log level should be set to debug.

Gloomyeye, 2025-05-02:
So far, I have the folloiwng installed:

- Sigen Hybrid 10.0 TP
- SigenSTor BAT 8.0 TP
- Sigen Sensor TP-CT-DH
- Hoymiles HM 600

I think that the energy of my Hoymiles HM 600 inverter is the problem. This adds more pv_power/grid_export.

TypQxQ, 2025-05-03:
Please see new error messages of v.1.0.6 for more detail and post here. Thank you!

Gloomyeye, 2025-05-03:
This is now the log with 1.0.6:

```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.sigen.calculated_sensor
Quelle: custom_components/sigen/calculated_sensor.py:347
Integration: Sigenergy ESS (Dokumentation, Probleme)
Erstmals aufgetreten: 12:10:44 (527 Vorkommnisse)
Zuletzt protokolliert: 13:08:39

[CS][Plant Consumed] Calculated power is negative: -0.1299999999999999 kW = 7.291 + 0 - 7.421 - 0.0
[CS][Plant Consumed] Calculated power is negative: -0.12799999999999923 kW = 7.291 + 0 - 7.419 - 0.0
[CS][Plant Consumed] Calculated power is negative: -0.1299999999999999 kW = 7.28 + 0 - 7.41 - 0.0
[CS][Plant Consumed] Calculated power is negative: -0.15500000000000025 kW = 7.282 + 0 - 7.437 - 0.0
[CS][Plant Consumed] Calculated power is negative: -0.14100000000000001 kW = 7.28 + 0 - 7.421 - 0.0
```

Wartopia, 2025-05-03:
I have this as well. I use a SolarEdge 12.5K on the smartport of the gateway.

I can see the inverter in the Sigen app but not in HA. 

Is there a way to add the Sigen Gateway HomeMax TP (3-fase)?

So it can read the solar by Sigen? Or is there a way i can add my own solar stats i get from the SolarEdge plugin in HA, this is also modbus.

Like a fake 4 or 5th PV systeem where we can add our own sensor data from a third party inverter?

TypQxQ, 2025-05-03:
This confirms that it is related to https://github.com/TypQxQ/Sigenergy-Local-Modbus/discussions/52 .
When the new Modbus parameters will be available I will integrate it into this.
Until then you can create a template where you add the "Sigen Plant Consumed" with the inverter calculation you have. Look here for more information: https://www.home-assistant.io/integrations/template/

I will close this because there is nothing I can do about it yet.

closed 2025-05-03 by TypQxQ

Wartopia, 2025-05-03:
I guess we need to wait then. I have zero clues how to make what you say. I have zero programming skills. I can replace stuff, not write it.

I know how to make a template in the UI, just not the steps to make the logic work or any code in it. I can make a daily kWh counter which resets each day, thats about it.

As far as i understand you, i can add my third party kWh sensor to the "Sigen Plant Consumed" one? Dont understand how that would help...

In the yaml version i replaced one of the PV plant sensors for kWh with one of my SolarEdge kWh sensors, so it did add automatically to the totaal of the sigen sensor produced.

If knew how to do that, all would keep working, just need a way to send kWh data to the totaal sensor of the sigen PV.

Any tips how i could do that?