TypQxQ/Sigenergy-Local-Modbus #72: DC charger start/stop not working

runn8, 2025-05-13:
[image]

discussion: [Link](https://github.com/TypQxQ/Sigenergy-Local-Modbus/discussions/70)

TypQxQ, 2025-05-13:
Thank you. You are probably the first who has one to test :)
I will review the code for it when I get back home.

runn8, 2025-05-19:
Is there maybe something I can do in the meantime?

RazorbackOz, 2025-05-23:
Hi there typQXQ,
recently got my DC charger and got the same issue as runn8.
Did some fiddling with the code and have got it to the point of attempting the write to Sigen.
I'm currently getting lost down this rabbit hole and it's been a long time since I've done any programming, So am hoping you can point me in the right direction.
i've attached a debug  log file.  anything starting with 'RBS', are debug statements i've added.

[home-assistant.log](https://github.com/user-attachments/files/20404388/home-assistant.log)

RazorbackOz, 2025-05-30:
All good now. Got the charger turning on/off. All figures seem to be correct.
Still need to sort some things out...
1. recognizing if the car is actually plugged in. Throws an error if it isn't.
2. determining if the charger is going though it's startup sequence, the on/off button goes back to off until the startup check is complete.
Any Ideas ?
happy to share my changes with you, just not where to send them.

TypQxQ, 2025-05-30:
Hi there @RazorbackOz !
Thank you for taking the time to make the changes.
You haven't actually made a pull request to the main project yet but I see that you have made changes you have uploaded here: https://github.com/RazorbackOz/Sigenergy-Local-Modbus/commit/0f0f704e28e5281e2e581b5de8f31ec8842aa3c7

The way you uploaded the files unfortunately made them appear that you have changed everything in all files and I can't see what you changed directly. Most likely it's because of line ending incompatibility between Linux (Home Assistant) and the system you used. 

I will try to use some other diff tools to see what changes you have made and try to incorporate them 😄

RazorbackOz, 2025-05-30:
hi typQxQ,
Your welcome, i've really enjoyed getting my head back into some real code again.
The changes were made from HA Studio Code Server via a Ubuntu box. So not sure about the end of line characters.
yea, figured i messed it up when i couldn't compare your files with mine. no problem, I'll figure it out and upload them properly later.
Every thing i've changed is marked with "RBS". ignoring the debug statements, you'll find the changes.
[DC Charger] Vehicle battery voltage, it hovers in the 0 - 2.0 volt range when not charging (plugged and unplugged). when it starts charging, it jumps to 380-400v. so not much help. 
That said, In sensor.py - native_value function, 
raw_value = self.coordinator.data.get("inverters", {}).get(self._device_name, {}).get(self.entity_description.key) #RBS - may need to change to "dc_chargers"
Haven't had time to look at as yet. I suspect it may be the reason the battery voltage isn't dropping to zero when unplugged (just holding the last value? Y/N/Maybe).
the vehicle SOC is in a similar boat.

purcell-lab, 2025-06-04:
> What is the state of `[DC Charger] Vehicle battery voltage` when the car is charging, plugged in vs. not plugged in ?
> Can we get a state of "Plugged in" from that?

Ran some tests and battery voltage drops to 0 when the charging session ends but the car is still plugged in.

However SOC keeps reporting, in this case an 80% charge limit, whilst the car is still plugged in so that may be the choice for "PluggedIn" = true when SOC > 0. Although I suppose the car could be plugged in with 0% SOC, hopefully rarely...

[image]

runn8, 2025-06-14:
The start/stop functionality is working now after the last Sigenergy and Sigenergy-local-modbus update.