TypQxQ/Sigenergy-Local-Modbus #74: Correct Integration With Predbat

Evaske, 2025-05-14:
Hi,

Thanks for the work on this integration. I've just recently swapped over to this new integration instead of the older version and just wanted to check I was doing the Predbat integration correctly.

For predbat to talk to the system it is making use of `modbus.write_register` which worked fine with the old integration. With the new integration Home Assistant knows nothing of `modbus` and so to get it to work I had to add the following to my configuration.yaml:

```
modbus:
  - name: Sigen
    type: tcp
    host: 192.168.0.167
    port: 502
    switches:
      - name: DummySwitch
        address: 13
        write_type: coil
```

Obviously this integration is already using modbus in the background so I shouldn't in theory have needed to add that entry to my configuration.yaml. What is the method to actually talk to the battery using the new integration.

This is part of the current code that is being used by Predbat from their docs that was mainly aimed at the old integration:

```
- id: "automation_sigen_ess_max_discharging_limit_input_number_action"
        alias: "sigen ESS max discharging limit input number action"
        description: "Sigen ESS max discharging limit action for batpred"
        triggers:
          - trigger: state
            entity_id: input_number.discharge_rate
        action:
          - action: modbus.write_register
            data_template:
              hub: Sigen
              slave: 247
              address: 40034
              value:
                - >-
                  0
                - >-
                  {{ states('input_number.discharge_rate') |
                  round(0) | int }}
        mode: single
```

Without the modbus entry in to configuration.yaml, there is a home assistant log entry that modbus doesn't exist.

TypQxQ, 2025-06-13:
You shouldn't use both integrations together because the poll intervall would be to small and Sigenergy give errors.

TypQxQ, 2025-06-15:
Here is some configuration I have been able to write together.
Please evaluate and if it works, I'll make it more readily available inside the FAQ and send it to PredBat.

Home Assistant configuration.yaml additions:
```
template:
  - sensor:
    - name: "Sigen Plant Battery Power W"
      unique_id: sigen_plant_battery_power_w
      state: >
        {% set power = states('sensor.sigen_plant_battery_power') | float(0) %}
        {{ (power * -1000) | round(0) }}
      unit_of_measurement: "W"
      device_class: power
      state_class: measurement
    - name: "Sigen Plant Consumed Power W"
      unique_id: sigen_plant_consumed_power_w
      state: >
        {% set power = states('sensor.sigen_plant_consumed_power') | float(0) %}
        {{ (power * 1000) | round(0) }}
      unit_of_measurement: "W"
      device_class: power
      state_class: measurement
    - name: "Sigen Plant Grid Active Power W"
      unique_id: sigen_plant_grid_active_power_w
      state: >
        {% set power = states('sensor.sigen_plant_grid_active_power') | float(0) %}
        {{ (power * -1000) | round(0) }}
      unit_of_measurement: "W"
      device_class: power
      state_class: measurement
    - name: "Sigen Plant PV Power W"
      unique_id: sigen_plant_pv_power_w
      state: >
        {% set power = states('sensor.sigen_plant_pv_power') | float(0) %}
        {{ (power * 1000) | round(0) }}
      unit_of_measurement: "W"
      device_class: power
      state_class: measurement
    - name: "Sigen Plant ESS Rated Discharging Power"
      unique_id: sigen_plant_ess_rated_discharging_power_w
      state: >
        {% set power = states('sensor.sigen_plant_ess_rated_discharging_power') | float(0) %}
        {{ (power * 1000) | round(0) }}
      unit_of_measurement: "W"
      device_class: power
      state_class: measurement
    - name: "Sigen Plant ESS Rated Charging Power"
      unique_id: sigen_plant_ess_rated_charging_power_w
      state: >
        {% set power = states('sensor.sigen_plant_ess_rated_charging_power') | float(0) %}
        {{ (power * 1000) | round(0) }}
      unit_of_measurement: "W"
      device_class: power
      state_class: measurement

input_number:
  sigen_plant_ess_max_discharging_limit_w:
    name: Sigen Plant ESS Max Discharging Limit W
    min: 0
    max: 500000
    step: 1
    mode: box
  sigen_plant_ess_max_charging_limit_w:
    name: Sigen Plant ESS Max Charging Limit W
    min: 0
    max: 500000
    step: 1
    mode: box

[... 535 lines cut ...]
  #    name: 'Snug import/Fixed export'
  #    rates_import_octopus_url: 'https://api.octopus.energy/v1/products/SNUG-24-11-07/electricity-tariffs/E-1R-SNUG-24-11-07-{octopus_region}/standard-unit-rates/'
  #    rates_export_octopus_url: 'https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-{octopus_region}/standard-unit-rates/'

  # Alert feeds - customise to your country and the alert types, severity and keep value
  #alerts:
  #  url: "https://feeds.meteoalarm.org/feeds/meteoalarm-legacy-atom-united-kingdom"
  #  event: "(Amber|Yellow|Orange|Red).*(Wind|Snow|Fog|Rain|Thunderstorm|Avalanche|Frost|Heat|Coastal event|Flood|Forestfire|Ice|Low temperature|Storm|Tornado|Tsunami|Volcano|Wildfire)"
  #  severity: "Moderate|Severe|Extreme"
  #  certainty: "Possible|Likely|Expected"
  #  keep: 40

  # Watch list, a list of sensors to watch for changes and then update the plan if they change
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in
  # Only uncomment the items you actually have set up above in apps.yaml, of course you can add your own as well
  # Note those using +[] are lists that are appended to this list, whereas {} items are single items only
  #watch_list:
  #  - '{octopus_intelligent_slot}'
  #  - '{octopus_ready_time}'
  #  - '{octopus_charge_limit}'
  #  - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'
```

TypQxQ, 2025-08-03:
> Thank you for your help with this. Using the configs above (changed for UK), predbat returns the following error
> 
> `Warn: Validation of apps.yaml found configuration item 'inverter_limit_charge' element sensor.sigen_plant_ess_rated_charging_power_w returned value None`

Did you add the template as above? That is where you should check for errors.

awslyall, 2025-08-03:
Yes, I used the template and I have moved forwards a little.

`sigen_plant_ess_rated_discharging_power`
`sigen_plant_ess_rated_charging_power`

The above two entities were disabled, which I presume is default. They have now been enabled.

Looking in Developer Tools/Entities I have the following entities, with what look to be the correct **values** and one is *1000 of the other (kW vs W)

`sigen_plant_ess_rated_discharging_power` & `sigen_plant_ess_rated_discharging_power_2`
`sigen_plant_ess_rated_charging_power` & `sigen_plant_ess_rated_charging_power_w`

Note the __2_ instead of the __w_

I renamed the discharging power entity, restarted the home assistent and the charging entity is now fixed (I dont know why this fixed an unrelated entity). However thr e outcome is that apps.yaml is no longereporting any error.

Ill now start try and understand if Predbat is actualy controlling Sigenstore. Im know I making progress as the logs are now stating that Im writing to the invertor register

`Total inverter register writes now 4`

awslyall, 2025-08-03:
ok, 

My changes above really helped the PredBat Plan and it was alot more obvious that it was trying to import/export energy from the grid at appropiate times. (

There are have a further two changes my side and I think,,,, Im working, as Predbat just started to export all the energy in my battery (as I write this, Im at the point where I would be gain the most value from selling excess energy)

1) Removed some template config from configuration.yaml that was using the old template syntax
2) Enabled Entity "Remote EMS Control Mode"

Finally, I did come across the following setting that I didnt change, but it might be worth noting. I think the HAC integration defaults to setting the Sigen Plant in "Read Only" mode

a) Settings/Devices and Services/Sigen ESS/Sigen Plant
b) Click Cog
c) Reconfigure Sigen Plant (Click Submit)
d) Untick Read Only Mode