TypQxQ/Sigenergy-Local-Modbus #75: 3rd Party Inverter, SDM120CT

Pukkahq, 2025-05-18:
We have a 3rd party inverter this is monitored via Sigenergy SDM120CT, this connects to Sigenergy inverter on pin 15/16. 
I can see the 3rd party inverter in the app. Is they anyway to see the info in HomeAssistance?

StroppyOz, 2025-05-19:
This used to work via the integration but the latest firmware update disabled the ability for 3rd party inverter PV generation to be reported via modbus.  Sigen have mentioned that this will be remediated in a future firmware updated but have not, as yet, committed to a timeframe.  A workaround is to install the integration for your inverter and have the reporting use it's PV generation with the remainder coming from the Sigen integration.

Pukkahq, 2025-05-19:
I did read the note about the firmware problem. but thought this was different. 
The meter is connected to the Inverter and not the Gateway.

StroppyOz, 2025-05-20:
I suspect it's more to do with how the modbus protocol has been implemented for 3rd party inverters than whether it's through the gateway or an alternative device.  As I understand it, it was the protocol that was modified and it is this that removed the reporting of 3rd party inverter data via the previously published modbus registers.

TypQxQ, 2025-06-16:
Added a new sensor for Third-Party PV Power in https://github.com/TypQxQ/Sigenergy-Local-Modbus/tree/Dev-1.0.9
Can someone please try it out to see if it works? I have no Third Party inverter to try it with.

TypQxQ, 2025-06-17:
Great!
Is the "Third-Party..." sensor working too?

StroppyOz, 2025-06-17:
Sorry, I thought the third party arthouse was plant pv. Will come back to you

StroppyOz, 2025-06-17:
Yes, the third-party sensor is working. In my case, it's the same as Plant PV.

[image]

Pukkahq, 2025-06-18:
I am happy to test, what's the process to switch to the dev branch? I did look in HACS

TypQxQ, 2025-06-19:
You'd have to manually copy the updated files from the Dev branch.

StroppyOz, 2025-06-19:
Third-Party PV Power is definitely tracking the same as that in the app.

john-s-au, 2025-06-24:
I can also confirm that third party PV power is tracking accurately

[image]