TypQxQ/Sigenergy-Local-Modbus #101: Integration becoming unavailable after changing modes

purcell-lab, 2025-07-01:
I am seeing a fairly regular occurrence where the integration sensors all become unavailable on a regular basis.

It looks like it is occurring when changing to the dis/charge mode.

[image]

I will enable debug and see if I can capture some detailed commands.

frver, 2025-07-01:
Do you only need to restart the integration, or also home assistant, or do you need to powercycle the inverter?

Reason for asking is that I'm running into similar issues. At first I thought it was the wireless connection, then changed to a LAN connection which improved the situation. 

What I am doing is using the integration to switch to external EMS mode when I want to utilize the battery to power the home consumption. (At this stage I didn't install the gateway yet and rely on the P1 meter to measure power consumption and production to the grid). If the battery is full I switch EMS mode off, otherwise the PV power gets curtailed. (I'm still on a 5ct/kWh plan).
What I notice that during the early morning or late afternoon, the control loop (in node-red) frequently changing between EMS mode on and off. When that happens occassionaly the integration fails. I've been able to pinpoint it to the inverter not responding anymore to modbus tcp connections. It does still respond on ping (only on LAN it is supported). 

I've now relaxed the changing over speed beween EMS ON and OFF modes, and it has been stable for 1.5 days now.

I was also experimenting with sending direct modbus TCP commands from node-red in stead of via the integration, but it requires some significant rewrite since the integration doesn't seem to pick up the changes I do via modbus directly (I control the plant power).
It seems that somehow me writing continously (once every one or two seconds) and changing EMS on/off mode has some adverse effect ultimately causing the inverter to not connect anymore at all.
It's just like if I trigger an error once (by writing an invalid setting via the integration, or command is not accepted by the inverter), this triggers this problem or at least make it significantly worse.

Maybe it is caused by the integration opening up another modbus connection after an error causing too many connections to be open at inverter which causes the effect described. (Just a hunch, didn't have the time to decipher the integration code)

purcell-lab, 2025-07-01:
> Do you only need to restart the integration, or also home assistant, or do you need to powercycle the inverter?

It generally comes back online by itself.

I can continue to ping the gateway and port 502 is still open, but the integration is showing that attribute error.

Have you been able to enable diagnostics and get the logs for your error?  If so please post them here.

closed 2025-08-03 by TypQxQ