TypQxQ/Sigenergy-Local-Modbus #102: Download Diagnostics - Login attempt or request with invalid authentication

purcell-lab, 2025-07-01:
When I try to download diagnostics, it points to the local site URL:
https://nyy7p9y9vjwsocdbtb83vt5iuguvimk8.ui.nabu.casa/api/diagnostics/config_entry/01JWFG0FYPW44D06HEH49DEXVN

However, I get a failure and error message in the logs:

[image]

[image]

```logs
2025-07-01 12:45:15.203 DEBUG (MainThread) [custom_components.sigen] Fetching data for HIGH took 1.340 seconds
2025-07-01 12:45:15.204 DEBUG (MainThread) [custom_components.sigen] Finished fetching sigen_192.168.1.149_502 data in 1.340 seconds (success: True)
2025-07-01 12:45:15.204 WARNING (MainThread) [custom_components.sigen.calculated_sensor] [CS][Plant Consumed] Calculated power is negative: -11.698 kW = 0.0 + 0 - 16.015 - -4.317
2025-07-01 12:45:15.206 DEBUG (MainThread) [custom_components.sigen.switch] DC charger identifier: Sigen Inverter, dc_charger_output_power: -13.056
2025-07-01 12:45:17.101 WARNING (MainThread) [homeassistant.components.http.ban] Login attempt or request with invalid authentication from localhost (127.0.0.1). Requested URL: '/api/diagnostics/config_entry/01JWFG0FYPW44D06HEH49DEXVN?authSig=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.XXXXXXX.9gTgly1KKtbIz8u-_wAqcNVcmbuL8SaMvZ0dv3GzVGU'. (Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/138.0.0.0 Safari/537.36 Edg/138.0.0.0)
2025-07-01 12:45:19.863 DEBUG (MainThread) [custom_components.sigen.modbus] Reading 21 Plant registers. update_frequency is 4
2025-07-01 12:45:20.425 DEBUG (MainThread) [custom_components.sigen.modbus] Reading 40 Inverter registers. update_frequency is 4
2025-07-01 12:45:20.834 DEBUG (MainThread) [custom_components.sigen.coordinator] DC charger found for inverter Sigen Inverter
2025-07-01 12:45:20.834 DEBUG (MainThread) [custom_components.sigen.modbus] Reading 40 Inverter registers. update_frequency is 4
```

TypQxQ, 2025-07-01:
It looks like you use a thord party inverter. You can try the latest beta.

purcell-lab, 2025-07-01:
> It looks like you use a third party inverter. You can try the latest beta.

No 3rd party inverter.

15 kW Sigen Energy Controller
8 kWh Sigen Battery
25 kW DCEV module (with beta bidirectional firmware)

Happy to try the beta..

TypQxQ, 2025-07-01:
I'll have to recheck the consumption calculation to take this case into account. Until then the consumption will be wrong.
I will check this when I get back from vacation in a weeks time.

purcell-lab, 2025-07-01:
> I'll have to recheck the consumption calculation to take this case into account

Thanks, I have created a template sensor as a workaround with the new calculations.

purcell-lab, 2025-08-12:
The 'Download diagnostics' button is still not functional, I suspect it is pointing to an incorrect internal location.

Please reopen this issue as unresolved.

[image]

purcell-lab, 2025-08-14:
Thanks this issue is resolved with v.1.1.4

[sigen-01K2BTHFJCNVTZ5TQN72WVSZJS-Sigen Inverter DC Charger-4649e841018047acb80762fde377f7f6.json](https://github.com/user-attachments/files/21780929/sigen-01K2BTHFJCNVTZ5TQN72WVSZJS-Sigen.Inverter.DC.Charger-4649e841018047acb80762fde377f7f6.json)