TypQxQ/Sigenergy-Local-Modbus #107: How to Display Today's Revenue on HA Dashboard

Richie-Rogers, 2025-07-10:
Hi,
I'm trying to display on my HA dashboard the Energy Value (£) that I can see on the Sigen app/website.
There doesn't seem to be a direct mapping for any of the values.
I know the cost is supposed to be KWh x export rate in pence, but it doesn't add up.
For example:
"Daily Grid Export Energy" is 19.6 KWh and my export rate is 16.5p = £3.23.4

However, my "Total Revenue" on Sigen app is £4.93

Looking on the app under "Yield Today" I see 27.12KWh, so: 27.12 x 16.5p = £4.47 (much closer, but still 46p out)

If I calculate my KWh export from the £4.93 / £0.165 I get = 29.88KWh.
That's about 2.7KWh out.

So, unless I'm missing something somewhere, is there actually a way for me to get a "relatively" accurate total revenue for the day or not?

Thanks,
Richie

RedAnt79, 2025-07-15:
For me the following calculation works:
(Daily Grid Export Energy * export rate) + ((Daily Grid Import Energy + Daily Yield - Daily Grid Export Energy) * import rate)

Although i have no idea how battery charge/discharge is counted.

RedAnt79, 2025-07-16:
Have you set up Home Assistant's energy feature? it will create the sensors that you need.
https://www.home-assistant.io/docs/energy/