TypQxQ/Sigenergy-Local-Modbus #111: sensor.sigen_plant_consumed_power stopped working under 1.0.9

dennisroethig, 2025-07-13:
Hi,

thanks for all the work and the latest release (1.0.9). I ran the update earlier and noticed that `sensor.sigen_plant_consumed_power` stopped coming through. Just reverted to 1.0.8 and things seem to work fine again.

Couldn't spot a specific reference to it in the logs, but happy to retest/debug if it helps in any way and you can't reproduce it 😅

[image]

dennisroethig, 2025-07-13:
Just has a quick look again and it seems like there is one issue/warning:

[image]

Probably related to https://github.com/TypQxQ/Sigenergy-Local-Modbus/issues/110

Gloomyeye, 2025-07-14:
Can confirm the same. Also, the sensor "binary_sensor.sigen_plant_pv_generating" is not working.

Ralf18225, 2025-07-14:
I have the same problem. Sigen Plant PV Power and Sigen Plant Consumed Power no longer work under 1.0.9
Thanks for the great work. I have gone back to 1.0.8.

lhjt, 2025-07-14:
> Can confirm the same. Also, the sensor "binary_sensor.sigen_plant_pv_generating" is not working.

I am also experiencing this issue - with the detail that it reports it as `Unknown`. I have a 3rd-party inverter connected.

TypQxQ, 2025-07-14:
Please check the new .10 version. Should be fixed now 😄