TypQxQ/Sigenergy-Local-Modbus #114: Typo in strings.json

Si-GCG, 2025-07-14:
There is an s missing from must in the following line: 

        "description": "Update the connection details and scan intervals for this Plant.\n\nHigh Priority Scan Interval is used for Power and Energy sensors.\nAlarm Scan Interval is used for Alarm sensors is usually set to same as Medium.\nScan intervals mut be a multiple of the High Scan Interval.",

“mut” changed to “must”

      "plant_config": {
        "title": "Reconfigure Sigenergy Plant",
        "description": "Update the connection details and scan intervals for this Plant.\n\nHigh Priority Scan Interval is used for Power and Energy sensors.\nAlarm Scan Interval is used for Alarm sensors is usually set to same as Medium.\nScan intervals must be a multiple of the High Scan Interval.",
        "data": {
          "host": "Host (IP address)",
          "port": "Port",
          "read_only": "Read-Only Mode",
          "scan_interval_high": "High Priority Scan Interval (seconds)",
          "scan_interval_alarm": "Alarm Scan Interval (seconds)",
          "scan_interval_medium": "Medium Priority Scan Interval (seconds)",
          "scan_interval_low": "Low Priority Scan Interval (seconds)"
        },
        "data_description": {
          "read_only": "(Prevents Home Assistant from making changes to the Sigenergy system. Recommended for initial setup. Can be changed later in device settings.)"
        }
      },

Also thanks for your awesome integration!!

purcell-lab, 2025-07-24:
Good work, have you considered updating this to a "Pull request"?

Si-GCG, 2025-07-24:
> Good work, have you considered updating this to a "Pull request"?
Hi there, I am still finding my way around this stuff so not really confident enough to jump into the whole pull request, but I really love this integration so if I can help I will.

purcell-lab, 2025-07-24:
Understand, GitHub can be daunting, but you are 95% of the way there and it's a great way to contribute. 

I recommend you start simple like this one, text correction.

Make a fork of the code, that way you can't break anything here. Make your changes by editing the code and then submit as a PR.

In some ways the simple PRs are actually easier than raising an issue.

TypQxQ, 2025-08-03:
Thank you @Si-GCG ,
Please see the comments I left on the pull req.

PR #132 (2025-08-03): Update strings.json - spelling
Closes #114
closed 2025-08-03 by TypQxQ