TypQxQ/Sigenergy-Local-Modbus #139: EVAC `ac_charger_start_stop` switch `is_on_fn` not correct

andrewnicols, 2025-08-08:
This is a bit of a weird one, and I'll try my best to explain.

The `ac_charger_start_stop::is_on_fn` will currently report a false value for the current states:
- Initializing
- Fault
- Error
- Not Connected

However there are also several other states which should report a false value:
- Reserving - Charger & EV Not Ready
- Preparing - Charger Ready, EV Not Ready
- EV Ready, Charger Not Ready

When the car is plugged in and in one of these states, you currently have to toggle the state to off, and then back on again.

TypQxQ, 2025-08-16:
Please check the suggestions I made.

TypQxQ, 2025-09-15:
I added it to a Beta Relase so anyone can test it. v.1.1.10 Beta 1
https://github.com/TypQxQ/Sigenergy-Local-Modbus/releases/tag/v.1.1.10.b1
Also have a dicusion open for it here: https://github.com/TypQxQ/Sigenergy-Local-Modbus/discussions/210