TypQxQ/Sigenergy-Local-Modbus #141: EVDC Sensors Unknown on v1.1.0

lordsnipe1, 2025-08-10:
The latest build does not appear to read the status of my EVDC charger correctly, which was working with the previous build.

The switch which starts/stops the EVDC charger is always ON and all sensors are "Unknown", even when the charger is connected and charging.

[image]

This was working on the previous build where the switch would normally be OFF, go ON when charging, and be able to be turned OFF to stop charging. Sensor data was also populating on the previous build.

Looking through the home-assistant.log file, there isn't anything in there for sigen-modbus apart from the first time I tried to toggle the switch with nothing connected and it throw the error:
[core-ssh homeassistant]$ cat home-assistant.log | grep modbus
2025-08-10 08:26:54.170 WARNING (MainThread) [custom_components.sigen.modbus] Modbus write error for 192.168.1.107:502@1 (address 41000): ExceptionResponse(dev_id=1, function_code=134, exception_code=2). Marking connection as closed.
2025-08-10 08:26:54.170 ERROR (MainThread) [custom_components.sigen.modbus] Failed to write dc_charger parameter 'dc_charger_start_stop' (device: Sigen Inverter): Error writing register: Error writing register at address 41000: ExceptionResponse(dev_id=1, function_code=134, exception_code=2)

If the charger is running and I press the switch, charging stops and the switch toggles back to ON.

With the charger plugged in (but not charging), turning off the switch reverts back to ON but charging does not start.

JP-Ellis, 2025-08-10:
I can confirm this is also happening to me. The DC charger subcomponent is no longer available since v1.1.0; with the same error as above:

```log
Aug 10 13:16:34 asrock hass[1339015]: 2025-08-10 13:16:34.507 ERROR (MainThread) [custom_components.sigen.modbus] Failed to write dc_charger parameter 'dc_charger_start_stop' (device: Sigen Inverter): Error writing register: Error writing register at address 41000: ExceptionResponse(dev_id=1, function_code=134, exception_code=2)
Aug 10 13:16:34 asrock hass[1339015]: 2025-08-10 13:16:34.507 ERROR (MainThread) [custom_components.sigen.coordinator] Failed to write parameter dc_charger_start_stop to dc_charger 'Sigen Inverter': Error writing register: Error writing register at address 41000: ExceptionResponse(dev_id=1, function_code=134, exception_code=2)
```

TypQxQ, 2025-08-11:
Can you please check the new v.1.1.1?
Also please all a log.
I can't debug DC chargers myself because I don't have one.
/Andrei

JP-Ellis, 2025-08-11:
When I plug the car in tomorrow, I'll see about producing proper debug logs to assist.

TypQxQ, 2025-08-11:
Hi!
More updates where made for this in v.1.1.2
Please see if it works now.

purcell-lab, 2025-08-12:
Upgraded to v.1.1.13 (v.1.1.3) and the EVDC data seems to have come good.

[image]

lordsnipe1, 2025-08-14:
This is a filter of custom_components.sigen logs from a fresh HA restart with debug enabled:

[sigen_logs_v1.1.4.txt](https://github.com/user-attachments/files/21765531/sigen_logs_v1.1.4.txt)

RazorbackOz, 2025-08-15:
Hi All,
TypQxQ, the values seem to be vaporising between _async_update_data and the HA Database.  Not sure where to look now, let me know if you do.

added  "_LOGGER.debug("RBS-coordinator-106-coordinator-post Merged fetched, len: %s data: %s", len(self.data), self.data) "
to the end of _async_update_data, line 106

got this...
[home-assistant.log](https://github.com/user-attachments/files/21782530/home-assistant.log)

Result... (145 is sensor.sigen_inverter_dc_charger_dc_charger_vehicle_battery_voltage)
[image]

TypQxQ, 2025-08-15:
Thank you!
I've released a alpha version for evaluation with your log addition and some changes to DC naming hoping to get it working.
Please try v.1.1.6.a1 and report back. https://github.com/TypQxQ/Sigenergy-Local-Modbus/releases/tag/v.1.1.6.a1
/Andrei

purcell-lab, 2025-08-15:
> Thank you! I've released a alpha version for evaluation with 

Thanks for tagging alpha release, it makes it a lot easier to download for testing. 

[image]

TypQxQ, 2025-08-15:
Finally progress. 
Can you please attach logs and diagnostics?
Why would power be negative, are you discharging?
Current is 0 which is strange at non 0 power.

purcell-lab, 2025-08-15:
> Finally progress. Can you please attach logs and diagnostics? Why would power be negative, are you discharging? Current is 0 which is strange at non 0 power.

Yes discharging is the reason for negative power.
Current has always shown 0 for discharging, it's correct for charging.

Diagnostic attached.

[sigen-01K2BTHFJCNVTZ5TQN72WVSZJS-Sigen Inverter DC Charger-4649e841018047acb80762fde377f7f6-1.json](https://github.com/user-attachments/files/21794226/sigen-01K2BTHFJCNVTZ5TQN72WVSZJS-Sigen.Inverter.DC.Charger-4649e841018047acb80762fde377f7f6-1.json)

purcell-lab, 2025-08-15:
I may of spoken too soon, It would appear that the DCEV switch is functional.

I will continue to monitor but it would appear we have full functionality of the DCEV charger again with v.1.1.6.a1, thanks.

[image]

JP-Ellis, 2025-08-17:
Thank you for fixing this quickly 🚀