TypQxQ/Sigenergy-Local-Modbus #150: Missing sensors in 3.1.0

TypQxQ, 2025-08-11:
> Same here sensor.sigen_plant_sigen_pv_power showing 0,000 kW since 1.1.0 was working before as i use that sensor for some automations. Plant software is V100R001C21SPC100 updated 31/7. Sigenstor retrofit with 12kw inverter and 24kwh batteries.
> 
> PV1-2-3 unavaible on all entites on those sensors.
> Sigen inventor unit all sensors unavaible.
> Sigen Plant unit some sensors show but yeah for example sensor.sigen_plant_sigen_pv_power dont show anything. 

 _Originally posted by @Kingcool88 in [#146](https://github.com/TypQxQ/Sigenergy-Local-Modbus/issues/146#issuecomment-3174197541)_

Kingcool88, 2025-08-11:
> [@Kingcool88](https://github.com/Kingcool88) Please provide what sensors are not working and a debug log as mentioned in the FAQ section of the discussions. My system is also in Sweden so we get the same updates and it's working fine. Have you tried restarting HA?

Yes i have restarted several times. Uploaded a log now. Havent cleaned it from unrelated stuff but most in it was from this integration.

Also got this error.

Loggare: homeassistant.helpers.entity
Källa: helpers/entity.py:1201
Inträffade först: 12:36:28 (3 förekomster)
Senast loggade: 17:28:33

Updating state for sensor.sigen_plant_plant_running_state (<class 'custom_components.sigen.sensor.SigenergySensor'>) took 0.587 seconds. Please create a bug report at https://github.com/TypQxQ/Sigenergy-Local-Modbus/issues
Updating state for binary_sensor.tygelvagen_40_online (<class 'custom_components.easee.binary_sensor.ChargerBinarySensor'>) took 0.506 seconds. Please create a bug report at https://github.com/nordicopen/easee_hass/issues
Updating state for sensor.sigen_plant_daily_pv_energy (<class 'custom_components.sigen.sensor.SigenergySensor'>) took 2.544 seconds. Please create a bug report at https://github.com/TypQxQ/Sigenergy-Local-Modbus/issues

[sigen.log](https://github.com/user-attachments/files/21719707/sigen.log)

Kingcool88, 2025-08-11:
> Problem seems to come from your Inverter not beeing found. Did it work in v.1.1.10? What is the name of the entities in the inverter? for example sensor.sigen_inverter.....

All was working like a Sharm before 1.1.0, after 1.1.0 update I had these issues.
sensor.sigen_inverter_active_power
sensor.sigen_inverter_battery_power
sensor.sigen_inverter_pv_power 
And so on

TypQxQ, 2025-08-11:
Try Device ID: 2

If this doesn't work, supply the debug log from a fresh restart. Any connection errors are at the begining, not found in what you sent.

Do you have a AC Charger?
DC Charger?

Kingcool88, 2025-08-11:
sensor.sigen_inverter_daily_pv_energy still not working tho. Don't know if you read it but I also have retrofit.

Kingcool88, 2025-08-11:
> Your latest logs show that it takes your computer more than 20 seconds to probe the registers for the inverter and it times out. This is intertwined with lots of errors from your other components and is probably what makes it so slow at startup. I'll try to check for this.

Even if this issue only occured only after 1.1.0 update?

TypQxQ, 2025-08-11:
Yes. The 1.1 is a big update with different style of modbus reading and added registers.
Try the 1.1.3 version. It should now retry after one minute if it fails to probe the registers.

Kingcool88, 2025-08-11:
With 1.1.**13** the sensors that were 0.00 before were now unavailable, went back to 1.1.2

Kingcool88, 2025-08-12:
Went back to 1.0.10 and all is working again.

closed 2025-08-14 by TypQxQ