TypQxQ/Sigenergy-Local-Modbus #155: ValueError: Sensor sensor.sigen_plant_charge_cut_off_soc

purcell-lab, 2025-08-12:
After upgrade to v.1.1.13 (v.1.1.3):

I see the following 

```logs
2025-08-12 11:14:10.805 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 663, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: 'unknown'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 666, in state
    numerical_value = float(value)
ValueError: could not convert string to float: 'unknown'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 280, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 491, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 190, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 571, in _handle_coordinator_update
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1026, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1151, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1088, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1032, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 670, in state
    raise ValueError(
    ...<5 lines>...
    ) from err
ValueError: Sensor sensor.sigen_plant_charge_cut_off_soc has device class 'None', state class 'None' unit '%' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'unknown' (<class 'str'>)
```

TypQxQ, 2025-08-14:
Please add logs: https://github.com/TypQxQ/Sigenergy-Local-Modbus/discussions/54
And Debug info: Home Assistant → Settings → Devices & Services → Sigenergy → Download Diagnostics, then attach.

TypQxQ, 2025-08-22:
Closing as no info added.

closed 2025-08-22 by TypQxQ