TypQxQ/Sigenergy-Local-Modbus #156: Battery Charge/Discharge and PV Daily Energy Stats Unknown (version 1.1.3/1.1.13)

lordsnipe1, 2025-08-12:
The following sensors are showing Unknown when using the latest version:
sensor.sigen_plant_daily_pv_energy
sensor.sigen_plant_daily_battery_charge_energy
sensor.sigen_plant_daily_battery_discharge_energy

These were showing data values on v1.1.2.

jyfab, 2025-08-12:
I'm also on latest firmware (SPC110), and the first sensor you mention is also showing Unknown but the other two are reporting correct values.

Adze1502, 2025-08-12:
Same here... Reverting to 1.1.2 resolves the issue

lordsnipe1, 2025-08-14:
On 1.1.4, this is what is in debug logging for these 3 sensors for me:
2025-08-14 11:02:56.488 DEBUG (MainThread) [custom_components.sigen.calculated_sensor] [CS][Daily PV] Static sensors not yet initialized, skipping calculation for 'inverter_daily_pv_energy'
2025-08-14 11:02:56.488 DEBUG (MainThread) [custom_components.sigen.calculated_sensor] [CS][Daily Batt Charge] Static sensors not yet initialized, skipping calculation for 'inverter_ess_daily_charge_energy'
2025-08-14 11:02:56.488 DEBUG (MainThread) [custom_components.sigen.calculated_sensor] [CS][Daily Batt Discharge] Static sensors not yet initialized, skipping calculation for 'inverter_ess_daily_discharge_energy'

lordsnipe1, 2025-08-14:
My log is attached to the EVDC issue https://github.com/TypQxQ/Sigenergy-Local-Modbus/issues/141#issuecomment-3186643265

closed 2025-08-14 by TypQxQ