TypQxQ/Sigenergy-Local-Modbus #196: [Bug]

rob0101, 2025-09-04:
### Integration version

1.1.8

### Describe the bug

I cannot change the max grid export limit, I get this error..

[image]

Wondering if this is 1.1.8 related or something Sig has introduced.

### Checklist

- [x] I searched existing issues and discussions.
- [x] I attached logs and/or diagnostics.
- [x] This is not a support request; it describes a reproducible bug.

Roving-Ronin, 2025-09-04:
Running v1.1.8 and also updated to HA 2025.9.0 and can confirm I also get the same message if trying to set a different max export limit.

jyavenard, 2025-09-04:
1.1.8 is also broken with 2025.9.0, most entities were marked as "unavailable" for me. Had to restore from backup.

Roving-Ronin, 2025-09-04:
JYA looks to be a localised issue, I'm running (see screenshot) and the sensors all come up OK / have updating data. Just the bug above re changing limits.

[image]

rob0101, 2025-09-04:
I lost all the sensors with 2025.9.0 and 1.1.7

TypQxQ, 2025-09-04:
Please supply logs with all error reports. I can't reproduce your error. 
1.1.8 is needed for HA 2025.9 which uses a new version of the Pympdbus module. Breaking change was a parameter name change from slave to something else and I fixed it in 1.1.8 so ot choses what parameter to use based on installed version of pympdbus. If you supply logs i can check it when i get the time. 

We had two beta releases and it was reported working fine.

rob0101, 2025-09-04:
>If you supply logs i can check it when i get the time. 
> 

Where will I find the logs you need ?

I can try again later once the export price stops being negative.

jyavenard, 2025-09-04:
> Just got Modbus turned on about 2hrs ago and cannot get it working. I noticed this post re 'Slave'. I was watching the installer with the Modbus option and it shoed either 'Slave' or 'Host'.

if you only have one EC, it must be set to host

edje, 2025-09-04:
` ModbusClientMixin.write_registers() got an unexpected keyword argument 'slave'.`

Since yesterday I have the same issue... I did the core update.

TypQxQ, 2025-09-04:
Please try out the new Beta version v.1.1.9.b1
I missed moving to the new check on two places in the code.
The code now has to call the functions with different parameter names depending on the version of pyModbus library HomeAssistant uses.

edje, 2025-09-04:
How can I update to the beta version ? Thanks!

Crosenhain, 2025-09-04:
Just installed that release, and it seems to be working so far. Thanks!

edje, 2025-09-04:
Thanks it is working again 👍🏻

ClaireRovic, 2025-09-05:
Thanks working at my end - I think the system is still 'Slave'