TypQxQ/Sigenergy-Local-Modbus #206: [Bug] Updating state for sensor.sigen_inverter_b_c_line_voltage

Wartopia, 2025-09-07:
### Integration version

1.1.9

### Describe the bug

Logger: homeassistant.helpers.entity
Source: helpers/entity.py:1201
First occurred: 01:11:04 (1 occurrence)
Last logged: 01:11:04

Updating state for sensor.sigen_inverter_b_c_line_voltage (<class 'custom_components.sigen.sensor.SigenergySensor'>) took 1.709 seconds. Please create a bug report at https://github.com/TypQxQ/Sigenergy-Local-Modbus/issues

Since a few days. Not sure if it was 1.1.9 or the update for HA 9.1

Happens only 1/2 times in the n8.

### Logs

Updating state for sensor.sigen_inverter_b_c_line_voltage (<class 'custom_components.sigen.sensor.SigenergySensor'>) took 1.709 seconds. Please create a bug report at https://github.com/TypQxQ/Sigenergy-Local-Modbus/issues

### Diagnostics file

[sigen-01JSV4PYQ6M9VVYNEP4ABN17EM-Sigen Plant-931b883f41e38723d5d83e026890a93a.json](https://github.com/user-attachments/files/22194146/sigen-01JSV4PYQ6M9VVYNEP4ABN17EM-Sigen.Plant-931b883f41e38723d5d83e026890a93a.json)

### Checklist

- [x] I searched existing issues and discussions.
- [x] I attached logs and/or diagnostics.
- [x] This is not a support request; it describes a reproducible bug.

TypQxQ, 2025-09-07:
Please attach your logs as described when creating the issue. There is no "sigen_inverter_b_c_line_voltage". Not even your diagnostics file contain that naming.

Wartopia, 2025-09-07:
Because its in the wrong place xD i did not see this. Sorry, ill do it. 5mins.

Place it after Logs: >>>>

Not under the log text window, because now its after the action taken. Thats why i missed it.

Wartopia, 2025-09-07:
I cant "Reproduce the Issue" because it happend when i sleep at 1AM.

not sure what you want me to do then.....

Wartopia, 2025-09-07:
Thats where the log comes from, there is nothing there ;(

I will open the diag from the sigen before i go to sleep and close it when i wake up. Hope it will happen again, it was 2 days now that i noticed the error in the log.

TypQxQ, 2025-09-08:
Great! We can always reopen it if you get it again.